openwrt/docker #7: Unable to find user root: no matching entries in passwd file

oxr463, 2019-07-11:
```sh
docker run --rm -it openwrtorg/rootfs:x86-64
Unable to find image 'openwrtorg/rootfs:x86-64' locally
x86-64: Pulling from openwrtorg/rootfs
90126f32b86a: Pull complete
Digest: sha256:d48323467d4813be476105dd80cc58fd9d7ca613de80ecfcc758aa4f414269c3
Status: Downloaded newer image for openwrtorg/rootfs:x86-64
docker: Error response from daemon: linux spec user: unable to find user root: no matching entries in passwd file.
```

aparcar, 2019-07-11:
*use docker they said, it works everywhere the same!*

```
a@oldschool:~$ sudo docker run --rm -it openwrtorg/rootfs:x86-64
[sudo] Passwort für a: 
Failed to resize receive buffer: Operation not permitted
/etc/preinit: line 6: can't create /sys/devices/system/cpu/microcode/reload: Read-only file system
ip: RTNETLINK answers: Operation not permitted
Press the [f] key and hit [enter] to enter failsafe mode
Press the [1], [2], [3] or [4] key and hit [enter] to select the debug level
ip: can't send flush request: Operation not permitted
ip: SIOCSIFFLAGS: Operation not permitted
Please press Enter to activate this console.
BusyBox v1.31.0 () built-in shell (ash)
/bin/ash: can't access tty; job control turned off
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r10451-653e05d27f
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@9d52013d3b7b:/# 
```

aparcar, 2019-07-11:
All right, after another pull I'm having the same issue, will investigate.

oxr463, 2019-07-11:
What is your host system?

I am running Ubuntu 18.04 in a virtual machine.

```sh
uname -a
Linux hostname 4.15.0-54-generic #58-Ubuntu SMP Mon Jun 24 10:55:24 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

cat /etc/os-release
NAME="Ubuntu"
VERSION="18.04.2 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.2 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic

docker -v
Docker version 18.09.7, build 2d0083d
```

aparcar, 2019-07-12:
We should fix this before proposing anything official. I wonder how this changed within the last days as it worked initially.

oxr463, 2019-07-12:
```sh
docker run --rm --user 0 -it openwrtorg/rootfs:x86-64
docker: Error response from daemon: OCI runtime create failed: container_linux.go:345: starting container process caused "exec: \"/sbin/init\": stat /sbin/init: no such file or directory": unknown.
```

Resources:

- https://docs.docker.com/engine/reference/run/#user

oxr463, 2019-07-12:
```sh
# Extract docker image
docker save -o openwrt_rootfs-x86-64.tar openwrtorg/rootfs:x86-64

# View contents of exported image
tar tvf openwrt_rootfs-x86-64.tar.gz
drwxr-xr-x 0/0               0 2019-07-11 00:01 0f8bc4e9d64a0431958b3e18b862ef2bed9b67fb7d1a99db7926f1c254e02cb0/
-rw-r--r-- 0/0               3 2019-07-11 00:01 0f8bc4e9d64a0431958b3e18b862ef2bed9b67fb7d1a99db7926f1c254e02cb0/VERSION
-rw-r--r-- 0/0            1359 2019-07-11 00:01 0f8bc4e9d64a0431958b3e18b862ef2bed9b67fb7d1a99db7926f1c254e02cb0/json
-rw-r--r-- 0/0            4096 2019-07-11 00:01 0f8bc4e9d64a0431958b3e18b862ef2bed9b67fb7d1a99db7926f1c254e02cb0/layer.tar
-rw-r--r-- 0/0            2570 2019-07-11 00:01 77ac5596b4abc092c0419ba2c2c4494c8f0680bf18611ada538dfb242ffb8538.json
-rw-r--r-- 0/0             213 1970-01-01 00:00 manifest.json
-rw-r--r-- 0/0             100 1970-01-01 00:00 repositories

# View file size of extracted image
ls -lh openwrt_rootfs-x86-64.tar.gz
-rw------- 1 user group 15K Jul 12 12:21 openwrt_rootfs-x86-64.tar.gz
```

oxr463, 2019-07-13:
I found the issue! Also, please check the email you have posted on your user profile. I sent you a very important message regarding a security issue.

```sh
#!/bin/bash -eo pipefail
bash docker-rootfs.sh

[ . . . TRUNCATED . . . ]

openwrt-x86-64-generic-rootfs.tar.gz

sent 43 bytes  received 3,347,102 bytes  1,338,858.00 bytes/sec
total size is 3,346,165  speedup is 1.00
openwrt-x86-64-generic-rootfs.tar.gz: OK
tar (child): openwrt-*-rootfs.tar.gz: Cannot open: No such file or directory
tar (child): Error is not recoverable: exiting now
tar: Child returned status 2
tar: Error is not recoverable: exiting now
```

Even though extracting the tarball is failing, the job still succeeds.

aparcar, 2019-07-13:
Hi @oxr463 , I fixed the problem by removing braces, which kill the file detection for tar. Also honored your mail, very professional, thanks!

closed 2019-07-13 by aparcar

PavelSosin-320, 2020-04-12:
I have the same when playing with another kind of Docker container for openWRT. Even when I run init manually I have it:
[image]
Maybe it's better to avoid using init for container initialization if it tends to fall into the interactive mode?  Let start container with the only Dropbear running and give developers to play with it using suspend, ssh, unsuspend, etc. ... and, even, debug using VSCode DAP.

oxr463, 2020-04-12:
> I have the same when playing with another kind of Docker container for openWRT. Even when I run init manually I have it:
> [image]
> Maybe it's better to avoid using init for container initialization if [...]

I would open another issue for this.
Also, normally init is not used in containers.

See: https://www.excella.com/insights/application-vs-system-containers

PavelSosin-320, 2020-04-13:
Let start Container in the "safe mode"