openwrt/docker #14: Building a 19.07 firmware

fantom-x, 2019-08-03:
Can you clarify in the readme how to make a 19.07 firmware? I just tried it the image builder is from master.

aparcar, 2019-08-03:
So far there are no public 19.07 binaries yet, if they appear I'd add them to the CI.

fantom-x, 2019-08-03:
But the 19.07 snapshot packages are already being built: https://downloads.openwrt.org/releases/19.07-SNAPSHOT/targets/. And I have been using them for a while.

aparcar, 2019-08-15:
Please try these new tags:
https://cloud.docker.com/u/openwrtorg/repository/docker/openwrtorg/rootfs/tags

fantom-x, 2019-08-15:
That link redirects me to https://hub.docker.com

fantom-x, 2019-08-15:
Oh, those are rootfs images. I was looking for "imagebuilder" for 19.07-SNAPSHOT...

fantom-x, 2019-08-15:
Is there a plan to make 19.07 imagebuilder images?

aparcar, 2019-08-15:
> Is there a plan to make 19.07 imagebuilder images?

Yes! Sorry right now I'm somewhat troubled with runners aka machines that build snapshot images on a daily basis. I'll try to figure it out asap and give you an update, sorry for the delay!

fantom-x, 2019-08-15:
Thx, looking forward to a faster and more convenient way to whip up a firmware image!

aparcar, 2019-08-16:
The 19.07-SNAPSHOT script now runs weekly, please try if it works for you!

fantom-x, 2019-08-16:
Thank you! This is great and simplifies/speeds up the process a lot!

I have built my images, but have not deployed them yet. One issue that I hit so far is no root access. One of my scripts is using **curl**, which is not present and I could not install it because there is no **sudo** nor there is a published root password that I could find (I did try **openwrt** though). 

I have updated my script to use **wget** instead, but I do see some uses for root access and having **sudo** would come really handy. Would you consider adding it?

aparcar, 2019-08-17:
Done via https://github.com/openwrt/docker/pull/25 please try it locally.

fantom-x, 2019-08-17:
Is it as simple as running docker-imagebuilder.sh ?

aparcar, 2019-08-17:
I'll update the README.md

* ./docker-common.sh to have the required gpg and usign keys
* export BRANCH="19.07-SNAPSHOT"
* sudo ./docker-imagebuilder.sh

fantom-x, 2019-08-17:
Ok, this is not working very well for me. First, **signify** was missing so I installed it (can it be added to the common script?).
Then, **signify (signify_1.14-3_all.deb)** did not like **-V**. I am using a fully up to date **Ubuntu 18.04.3 LTS**

```
+ '[' -f sha256sums.sig ']'
+ hash signify-openbsd
+ SIGNIFY_BIN=signify
+ VERIFIED=
+ for KEY in ./usign/*
+ echo 'Trying ./usign/b5043e70f9a75cde...'
Trying ./usign/b5043e70f9a75cde...
+ signify -V -q -p ./usign/b5043e70f9a75cde -x sha256sums.sig -m sha256sums
Signify: Unknown parameter '-V'
+ for KEY in ./usign/*
+ echo 'Trying ./usign/f94b9dd6febac963...'
Trying ./usign/f94b9dd6febac963...
+ signify -V -q -p ./usign/f94b9dd6febac963 -x sha256sums.sig -m sha256sums
Signify: Unknown parameter '-V'
+ '[' -z '' ']'
+ echo 'Could not verify usign signature'
Could not verify usign signature
+ exit 1
+ exit 1
```

fantom-x, 2019-08-17:
Ah, it should have been **signify-openbsd** (I removed **signify** first). Another small comment, the **docker-imagebuilder.sh** does not warn if it is run as a non-root user.

fantom-x, 2019-08-17:
Your fix is working: I tested by running **sudo apt update** and other commands. Thank you.

Three minor hick-ups:

1. **signify-openbsd** needs to be installed first on Ubuntu
2. **docker-imagebuilder.sh** does not warn about being run as a non-root user
3. The resulting docker image is not fully patched (see the last line in the output below). Maybe the Docker file could run **apt dist-upgrade** to always generate up-to date image?

```
sudo apt update     
Get:1 http://security-cdn.debian.org/debian-security buster/updates InRelease [39.1 kB]
Get:2 http://cdn-fastly.deb.debian.org/debian buster InRelease [118 kB]             
Get:4 http://security-cdn.debian.org/debian-security buster/updates/main amd64 Packages [62.9 kB]
Get:3 http://cdn-fastly.deb.debian.org/debian buster-updates InRelease [49.3 kB]
Get:5 http://cdn-fastly.deb.debian.org/debian buster/main amd64 Packages [7897 kB]
Get:6 http://cdn-fastly.deb.debian.org/debian buster-updates/main amd64 Packages [884 B]                                                    
Fetched 8167 kB in 7s (1089 kB/s)                                                                                                           
Reading package lists... Done
Building dependency tree       
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.

```

aparcar, 2019-08-17:
Hi, thanks for testing, I'd be happy if you could create some PRs for that.

aparcar, 2019-08-19:
I'm thinking of replacing the docker-* scripts with multi stage builds in the Dockerfile, will see...

aparcar, 2020-05-20:
I never replaced it with the multi stage builds, works *somewhat* stable lately. @fantom-x if you want more docs on running it outside the CI please create a PR, I'll close this for now.

closed 2020-05-20 by aparcar