openwrt/docker #28: docker cant run openwrt 1806.4 rootfs.tar.gz

siasjack, 2019-09-26:
hi
I am jack from china shanghai,I am new for docker ,i want run openwrt 1806.4 rootfs as a image of docker.

I run this cmd:  
`
docker import openwrt-x86-64-generic-rootfs.tar.gz
docker run -d --name='op3' c4d3107679c2 /sbin/init      #docker start ok,but run ps find  uhcpc uhttpd netifd etc.. is not running!

docker run -d --cap-add=NET_ADMIN --name='op3'  c4d3107679c2 /sbin/init   #openwrt running fine,all  important process working!

  
`
I find @zoobab(http://www.zoobab.com/docker-openwrt-image) image(openwrt version is AA too old) work fine,how??
I want to build a openwrt 1806.4 rootfs,whick running in  docker no need net_admin or privileged,pls help!

thanks very much!!

aparcar, 2019-09-28:
I'm guessing it stuck because the rootfs tries to setup the LAN interface, which it can't without NET_ADMIN. To tackle this I added a [config](https://github.com/openwrt/docker/blob/master/rootfs/etc/config/network) to the rootfs container, which isn't done when you use `docker import`.

If you want to do local testing within docker please see this proposed script https://github.com/openwrt/openwrt/pull/2367

oxr463, 2020-03-17:
We should document importing and running images in the README.md file.

aparcar, 2023-04-18:
This is old, closing it.

closed 2023-04-18 by aparcar