openwrt/docker #43: mkdir: cannot create directory '/home/build/openwrt/bin/targets': Permission denied

oxr463, 2020-02-19:
```sh
mkdir -p public
docker run --rm -v "$(pwd)"/public/:/home/build/openwrt/bin openwrtorg/imagebuilder make image PACKAGES="wifidog"
Unable to find image 'openwrtorg/imagebuilder:latest' locally
latest: Pulling from openwrtorg/imagebuilder
dc65f448a2e2: Pulling fs layer
974fa924b194: Pulling fs layer
7ca3655cacfd: Pulling fs layer
46820d6b3cc1: Pulling fs layer
37f28a662628: Pulling fs layer
395f379379c5: Pulling fs layer
46820d6b3cc1: Waiting
37f28a662628: Waiting
395f379379c5: Waiting
7ca3655cacfd: Verifying Checksum
7ca3655cacfd: Download complete
dc65f448a2e2: Verifying Checksum
dc65f448a2e2: Download complete
46820d6b3cc1: Verifying Checksum
46820d6b3cc1: Download complete
395f379379c5: Verifying Checksum
395f379379c5: Download complete
37f28a662628: Verifying Checksum
37f28a662628: Download complete
974fa924b194: Verifying Checksum
974fa924b194: Download complete
dc65f448a2e2: Pull complete
974fa924b194: Pull complete
7ca3655cacfd: Pull complete
46820d6b3cc1: Pull complete
37f28a662628: Pull complete
395f379379c5: Pull complete
Digest: sha256:355c90f9d3c58a4452cf15daa577303f0ca62c4c67fa40c0cebdb1d667da3e11
Status: Downloaded newer image for openwrtorg/imagebuilder:latest
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'gcc'... ok.
Checking 'working-gcc'... ok.
Checking 'g++'... ok.
Checking 'working-g++'... ok.
Checking 'ncurses'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'getopt'... ok.
Checking 'stat'... ok.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'perl'... ok.
Checking 'python2-cleanup'... ok.
Checking 'python'... ok.
Checking 'python3'... ok.
Checking 'git'... ok.
Checking 'file'... ok.
Checking 'ldconfig-stub'... ok.
Building images for x86 - Generic x86/64
Packages: wifidog base-files busybox dnsmasq dropbear e2fsprogs firewall fstools ip6tables iptables kernel kmod-bnx2 kmod-button-hotplug kmod-e1000 kmod-e1000e kmod-igb kmod-ipt-offload kmod-r8169 libc libgcc logd mkf2fs mtd netifd odhcp6c odhcpd-ipv6only opkg partx-utils ppp ppp-mod-pppoe uci uclient-fetch urandom-seed urngd
mkdir: cannot create directory '/home/build/openwrt/bin/targets': Permission denied
make[1]: *** [Makefile:116: _call_image] Error 1
make: *** [Makefile:197: image] Error 2
```

oxr463, 2020-02-21:
I solved this by changing the directory owner via `chown -R 1000:1000 public`.

aparcar, 2020-03-26:
Please close this if the issue if resolved.

closed 2020-03-26 by oxr463