openwrt/docker #47: Exiting from top causes container to shut down

lePereT, 2020-04-13:
For the previous issue I used `top` inside the container to determine that something was up with `urngd`. However, consistently when leaving `top` with a `ctrl + c` it causes the container itself to crash.

To replicate:
1) `docker run --rm -it openwrtorg/rootfs`
2) `top`
3) `ctrl + c`
4) wait 2 seconds

lePereT, 2020-04-13:
On some further digging, I think it's related to the errror message that appears just above the banner on login using the Readme command of `docker run --rm -it openwrtorg/rootfs`, which is `/bin/ash: can't access tty; job control turned off`

This looks like it means that advanced tty functionality like `ctrl + c` is unavailable

lePereT, 2020-04-13:
So this gets weirder, and it's now exceeded my capacity to diagnose. Using the readme's command `docker run --rm -it openwrtorg/rootfs` leads to the issue here and the previous one I posted in #46. 

However, if I use `docker run -it openwrtorg/rootfs` it again displays the same problems. But if I restart the crashed container then it behaves seemingly perfectly. Any idea as to what's happening here?

aparcar, 2020-05-20:
I'm aware of the ctrl + c (SIGINT) issue, however don't know how to fix it. If you come up with an idea please help me out.

@lynxis do you have an idea regarding the *tty access trouble*?