openwrt/docker #48: Docker cant run image openwrtorg/rootfs:x86-64

ZoZhang, 2020-05-05:
#### Error by docker build
````
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-mod-admin-full:
 * 	kernel (= 5.4.38-1-9dea1f6ed59a3231b5bc094092a9414a)
 * opkg_install_cmd: Cannot install package luci-mod-admin-full.
````

#### Dockerfile 
```
FROM openwrtorg/rootfs:x86-64

MAINTAINER Zhao ZHANG <zo.zhang@gmail.com>

USER root

RUN mkdir /var/lock && \
    opkg update && \
    opkg install uhttpd luci-base luci-mod-admin-full luci-theme-bootstrap luci-i18n-base-en luci-i18n-firewall-en luci-i18n-base-fr luci-i18n-firewall-fr && \
    uci set luci.ccache.enable=0 && \
    uci commit luci

EXPOSE 80
```

aparcar, 2020-05-05:
Can you please run `uname -a` in the docker container? I'm wondering if it is because the host system uses a different kernel than opkg excepts, but that's may be unrelated. I'll keep investigating

jonglezb, 2020-05-05:
Could it be that the rootfs snapshot of the docker image is too old for the current luci snapshot from downloads.openwrt.org?

I'm actually surprised that luci depends on the kernel package, but the error looks like the common version mismatch that happens when using older snapshots and installing newer packages.

A workaround is to use `opkg --force-depends` to ignore the dependency, but I guess the proper fix would be to rebuild the rootfs docker image regularly.

ynezz, 2020-05-06:
> but I guess the proper fix would be to rebuild the rootfs docker image regularly.

They're build every day at 4am via GitLab [pipeline](https://gitlab.com/openwrt/docker/pipelines/143152842) [schedules](https://gitlab.com/openwrt/docker/pipeline_schedules). Currently it's broken as one of the tests for image builder is currently [failing]](https://gitlab.com/openwrt/docker/-/jobs/540067478) due to the x86/64 image generation refactoring ` ls: cannot access './bin/targets/x86/64/*combined-squashfs.img.gz': No such file or directory` and we're simply not aware about it.

It's quite strange, that GitLab doesn't send out failure emails in case of scheduled pipeline failures like it does for ordinary after the `git push` pipeline runs.

ynezz, 2020-05-06:
> It's quite strange, that GitLab doesn't send out failure emails in case of scheduled pipeline failures like it does for ordinary after the git push pipeline runs.

Seems like this needs to be explicitly configured in https://gitlab.com/openwrt/docker/-/services/pipelines_email/edit

aparcar, 2020-05-06:
GitHub automatically closed this issue by merging PR #50, please reopen if the issue is still existing...

closed 2020-05-06 by aparcar

ynezz, 2020-05-06:
I've just tested it and the `opkg install uhttpd luci-base luci-mod-admin-full luci-theme-bootstrap luci-i18n-base-en luci-i18n-firewall-en luci-i18n-base-fr luci-i18n-firewall-fr` worked.

PR #50 (2020-05-06): ci: fix imagebuilder test for x86/64
That check is currently failing due to x86 image generation code
refactoring which changed naming of resulting images.

 ls: cannot access './bin/targets/x86/64/*combined-squashfs.img.gz': No such file or directory

Fixes #48
commit a7c9b692 (2020-05-06): ci: fix imagebuilder test for x86/64

That check is currently failing due to x86 image generation code
refactoring which changed naming of resulting images.

 ls: cannot access './bin/targets/x86/64/*combined-squashfs.img.gz': No such file or directory

Fixes #48