openwrt/docker #52: Can't check gpg signature when building rootfs

jonglezb, 2020-05-20:
I tried building a rootfs for the latest 19.07 release with:
 
    $ BRANCH=19.07.3 ./docker-rootfs.sh

However it failed to check the gpg signature:

```
+ export GNUPGHOME=/keys/gpg/
+ GNUPGHOME=/keys/gpg/
+ export USIGNHOME=/keys/usign/
+ USIGNHOME=/keys/usign/
+ curl https://downloads.openwrt.org/releases/19.07.3/targets/x86/64/sha256sums -fs -o sha256sums
+ curl https://downloads.openwrt.org/releases/19.07.3/targets/x86/64/sha256sums.asc -fs -o sha256sums.asc
+ curl https://downloads.openwrt.org/releases/19.07.3/targets/x86/64/sha256sums.sig -fs -o sha256sums.sig
+ '[' '!' -f sha256sums.asc ']'
+ '[' '!' -f sha256sums.asc ']'
+ gpg --with-fingerprint --verify sha256sums.asc sha256sums
gpg: keyblock resource `/keys/gpg//pubring.gpg': file open error
gpg: Signature made Sun 17 May 2020 02:42:57 AM CEST using RSA key ID 2074BE7A
gpg: Can't check signature: public key not found
+ exit 1
```

Are those `/keys/` paths supposed to be on the host system? Obviously it doesn't work in my case because I don't run the script as root...

aparcar, 2020-05-20:
Please test https://github.com/openwrt/docker/pull/55 and set GNUPGHOME and USIGNHOME to something accessible by your current user.

Also you have to run `docker-common.sh` first to download the required keys.

jonglezb, 2020-05-21:
Thanks, that worked, except that `USIGNHOME` needs a trailing slash otherwise this part fails:

```
+ for KEY in '"$USIGNHOME"*'
+ echo 'Trying keys/usign...'
```

Is there anything preventing using local dirs (`keys/gpg/` and `keys/usign/`) by default?

aparcar, 2020-05-21:
I think some folder switching I used absolute path, however please try and create a PR