openwrt/docker #82: How to expose LuCI outside container

oxr463, 2021-05-22:
I have a Dockerfile [[1]] that installs `luci` and `luci-ssl`, and a docker-compose.yml file [[2]], that exposes ports 80 and 443, but when I visit http://localhost or https://localhost in my browser, it isn't showing up. Also, I don't see any logs from uhttpd.

[1]: https://github.com/oxr463/luci-cli/blob/master/Dockerfile
[2]: https://github.com/oxr463/luci-cli/blob/master/docker-compose.yml

nono723, 2022-03-25:
Hi ^^

have you found a solution to your problem ?

thanks

oxr463, 2022-03-25:
No, I wasn't able to figure it out.

aparcar, 2022-03-25:
I'm able to access the interface just fine:

```
root@chef:~# curl https://localhost:4433 -k
<?xml version="1.0" encoding="utf-8"?>
<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.1//EN" "http://www.w3.org/TR/xhtml11/DTD/xhtml11.dtd">
<html xmlns="http://www.w3.org/1999/xhtml">
	<head>
		<meta http-equiv="Cache-Control" content="no-cache, no-store, must-revalidate" />
		<meta http-equiv="refresh" content="0; URL=cgi-bin/luci/" />
		<style type="text/css">
			body { background: white; font-family: arial, helvetica, sans-serif; }
			a { color: black; }

			@media (prefers-color-scheme: dark) {
				body { background: black; }
				a { color: white; }
			}
		</style>
	</head>
	<body>
		<a href="cgi-bin/luci/">LuCI - Lua Configuration Interface</a>
	</body>
</html>
```

The running container is based on your Dockerfile started via `docker run --rm -it -p 8080:80 -p 4433:443 rootfs-test`

oxr463, 2022-04-27:
This is what I'm getting:

```
docker run --rm -it -p 8080:80 -p 4433:443 oxr463/luci-cli
Failed to resize receive buffer: Operation not permitted
/etc/preinit: line 5: can't create /sys/devices/system/cpu/microcode/reload: Read-only file system
ip: RTNETLINK answers: Operation not permitted
Press the [f] key and hit [enter] to enter failsafe mode
Press the [1], [2], [3] or [4] key and hit [enter] to select the debug level
ip: can't send flush request: Operation not permitted
ip: SIOCSIFFLAGS: Operation not permitted
Please press Enter to activate this console.
```