openwrt/docker #83: Backporting for Chaos Calmer

oxr463, 2021-07-06:
I'm trying to create an sdk image for 15.05, but I am unsure how to create an SDK image manually from my terminal.

So far the only change I've made is downgrading the base image to Debian 8.

See: https://github.com/oxr463/openwrt-docker/tree/chaos-calmer

aparcar, 2021-07-06:
Please don't use CC anymore, it's probably easier to upgrade your stack to 19.07 or whatever else than it is to backport things.

oxr463, 2021-07-06:
> Please don't use CC anymore, it's probably easier to upgrade your stack to 19.07 or whatever else than it is to backport things.

I agree, that would be the right approach. But the build system has been customized too much and it would take considerable effort to upgrade.

It's definitely easier for me to modify a few Dockerfiles and build as-is.

aparcar, 2021-07-14:
I'll rework the documentation on how to run the scripts locally, however I'll not help setting up the build environment. 15.05 predates my OpenWrt involvement and I'm no fan of supporting the release with all it's security issues

oxr463, 2021-07-14:
Thank you. That would be extremely helpful.