openwrt/docker #88: Consider using OCI Image manifest fields like platform/arch/os instead of platform/arch tags

xavier83, 2021-09-14:
Current the use of tags for different architecture might not be the recommended way to distribute containers for different platforms/architecture/os. See https://docs.docker.com/engine/reference/commandline/manifest/#manifest-annotate

https://docs.docker.com/registry/spec/manifest-v2-2/
https://github.com/opencontainers/image-spec/blob/main/image-index.md

If enabled the container specification in docker-compose/kubernetes config won't have to worry about specifying the right architecture tag.

aparcar, 2022-04-19:
Hi, could you provide a PR for that?

aparcar, 2023-04-25:
I guess this is related to https://github.com/openwrt/docker/issues/112

lu-zero, 2024-02-16:
I tried with the current `nerdctl` and the problem still persists