openwrt/docker #91: example for gitlab-ci

stefangweichinger, 2022-02-25:
I am looking for an example how to use the image-builder docker image in a GitLab CI/CD pipeline.
Could anyone post one or add a link/pointer here? Thanks a lot.

aparcar, 2022-02-27:
https://github.com/openwrt/docker/blob/master/.gitlab/ci/imagebuilder.yml#L13

stefangweichinger, 2022-02-27:
As far as I understand the quoted yml builds the image-builder docker image, right?
I am looking for an example how to *use* that docker image inside a build pipeline to actually build openwrt-image-files.

Maybe I misunderstand, pls advise.

ynezz, 2022-02-27:
> As far as I understand the quoted yml builds the image-builder docker image, right?

No. See https://hub.docker.com/r/openwrtorg/imagebuilder or https://github.com/openwrt/docker#imagebuilder for another example, there is documentation as well https://openwrt.org/docs/guide-user/additional-software/imagebuilder. This is not a support channel, only bug reports are expected, for such basic support questions please use [forums](https://forum.openwrt.org/c/devel/8) instead.

closed 2022-02-27 by ynezz