openwrt/docker #93: Provide a guide for using the image

firstdorsal, 2022-05-03:
Hi, I found this repo and the docker image but there are no real instructions provided for running it.

Sure there is: `docker run --rm -it openwrtorg/rootfs` but this does not help as it does not get the container running properly:
```sh
Failed to resize receive buffer: Operation not permitted
/etc/preinit: line 5: can't create /sys/devices/system/cpu/microcode/reload: Read-only file system
ip: RTNETLINK answers: Operation not permitted
Press the [f] key and hit [enter] to enter failsafe mode
Press the [1], [2], [3] or [4] key and hit [enter] to select the debug level
ip: can't send flush request: Operation not permitted
ip: SIOCSIFFLAGS: Operation not permitted
Please press Enter to activate this console.
```

What caps does one need to add to get this running? 
Does it need to be in privileged mode?
What devices need to be mounted into the container?
How can the webinterface be launched?

I want to use this image for having a router on my home server in docker for my home network.

Thank you and all the best!

aparcar, 2022-05-04:
I merely used the rootfs for CI testing and not as a real deployment. If you have something specific in mind, please elaborate and we can see how to work things out.

Generally you can tell Docker to create multiple network interfaces for the router, on first boot OpenWrt will create by default a LAN interface (dhcp server) on eht0 and (if exists) a WAN interface (dhcp client) on eth1. Docker on the other hand automatically assigns IP addresses to container network interfaces. To make this work, you'd need to create an appropriate `/etc/config/network` file and mount it into the container.

Another step could be to improve the default configuration of the `rootfs` container, I did some things over [here](https://gitlab.com/openwrt/docker/-/merge_requests/47/diffs) but never followed up with it.

firstdorsal, 2022-05-04:
My plan was using openwrt as home router.

I am running arch linux on an always on server machine that I want to deploy other containers to.

My current home router is a bit slow though i was interested in trying out openwrt. As I already have an OS installed on the server the only solution i saw was virtualizing openwrt but that turned out to be quite difficult as i am no expert in using kvm nor have any interface other than SSH into the server.
So I found this container and thought this would be a much more approachable solution (also for the many people wanting to use openwrt on a raspberrypi) as i use docker all the time.
But as no further information is provided i don't know if the container has to run in privileged mode or what devices need to be passed through how to this image. Also i would love to use the webinterface for openwrt.

Thank you and all the best 
Paul

ES-giacomo, 2022-07-12:
> Hi, I found this repo and the docker image but there are no real instructions provided for running it.
> 
> Sure there is: `docker run --rm -it openwrtorg/rootfs` but this does not help as it does not get the container running properly:
> 
> ```shell
> Failed to resize receive buffer: Operation not permitted
> /etc/preinit: line 5: can't create /sys/devices/system/cpu/microcode/reload: Read-only file system
> ip: RTNETLINK answers: Operation not permitted
> Press the [f] key and hit [enter] to enter failsafe mode
> Press the [1], [2], [3] or [4] key and hit [enter] to select the debug level
> ip: can't send flush request: Operation not permitted
> ip: SIOCSIFFLAGS: Operation not permitted
> Please press Enter to activate this console.
> ```
> 
> What caps does one need to add to get this running? Does it need to be in privileged mode? What devices need to be mounted into the container? How can the webinterface be launched?
> 
> I want to use this image for having a router on my home server in docker for my home network.
> 
> Thank you and all the best!

Completely agree with @firstdorsal 
I'm also looking forward for a real instruction example to properly make a docker image

Special focus on arm target would be great since the market is doing a lot about those device (raspberry but also rockpi) and they are getting more and more network devices. An openwrt cointainer could easily be a must have to managed al these interfaces.

Thank you for your time and efford!

blackliner, 2022-11-27:
+1, would love to see OpenWRT being able to run as a container on a Raspi4. Currently all of its resources go to waste to run a 5Mb OS, and it would be good to put the other 3.9 cores and 3Gb RAM to some use.

aparcar, 2023-04-18:
Containers are somewhat meant to run a single tool and not an operating system including a process manager. Without services like ubusd OpenWrt is not fun. Isn't Alpine an option for you?

geryescalier, 2025-07-05:
> Containers are somewhat meant to run a single tool and not an operating system including a process manager. Without services like ubusd OpenWrt is not fun. Isn't Alpine an option for you?

Hi, do you have any info on this or a repo on how to implement it with Alpine? Thanks!