openwrt/docker #111: openwrt/imagebuilder:x86-64-v22.03.4 is building master....

fantom-x, 2023-04-22:
@aparcar 

```
REPOSITORY             TAG               IMAGE ID       CREATED        SIZE
openwrt/imagebuilder   x86-64-v22.03.4   09e27754d696   21 hours ago   1.06GB

============================================
cat repositories.conf 
## Place your custom repositories here, they must match the architecture and version.
# src/gz %n https://downloads.openwrt.org/snapshots
# src custom file:///usr/src/openwrt/bin/x86/packages

## Remote package repositories
src/gz openwrt_core https://downloads.openwrt.org/snapshots/targets/x86/64/packages
src/gz openwrt_base https://downloads.openwrt.org/snapshots/packages/x86_64/base
src/gz openwrt_kmods https://downloads.openwrt.org/snapshots/targets/x86/64/kmods/5.15.107-1-e469f5589b4c7b368924a6e4f8f7407f
src/gz openwrt_luci https://downloads.openwrt.org/snapshots/packages/x86_64/luci
src/gz openwrt_packages https://downloads.openwrt.org/snapshots/packages/x86_64/packages
src/gz openwrt_routing https://downloads.openwrt.org/snapshots/packages/x86_64/routing
src/gz openwrt_telephony https://downloads.openwrt.org/snapshots/packages/x86_64/telephony

## This is the local package repository, do not remove!
src imagebuilder file:packages

option check_signature

=========================================================
Building package index...
Downloading https://downloads.openwrt.org/snapshots/targets/x86/64/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/snapshots/targets/x86/64/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/x86_64/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/snapshots/packages/x86_64/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/targets/x86/64/kmods/5.15.107-1-e469f5589b4c7b368924a6e4f8f7407f/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_kmods
Downloading https://downloads.openwrt.org/snapshots/targets/x86/64/kmods/5.15.107-1-e469f5589b4c7b368924a6e4f8f7407f/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/x86_64/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/snapshots/packages/x86_64/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/x86_64/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/snapshots/packages/x86_64/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/x86_64/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/snapshots/packages/x86_64/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/x86_64/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/snapshots/packages/x86_64/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.

Installing packages...
Installing libc (1.2.3-4) to root...
Installing libgcc1 (12.2.0-4) to root...
Downloading https://downloads.openwrt.org/snapshots/targets/x86/64/packages/libgcc1_12.2.0-4_x86_64.ipk
Configuring libgcc1.
Configuring libc.
Installing kernel (5.15.107-1-e469f5589b4c7b368924a6e4f8f7407f) to root...
Configuring kernel.
Installing htop (3.2.2-1) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/x86_64/packages/htop_3.2.2-1_x86_64.ipk
Installing terminfo (6.4-2) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/x86_64/base/terminfo_6.4-2_x86_64.ipk
Installing libncurses6 (6.4-2) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/x86_64/base/libncurses6_6.4-2_x86_64.ipk

```

fantom-x, 2023-04-22:
And so is `openwrt/imagebuilder:x86-64-22.03.4-SNAPSHOT`: both are building master.

```
REPOSITORY             TAG                       IMAGE ID       CREATED        SIZE
openwrt/imagebuilder   x86-64-22.03.4-SNAPSHOT   09e27754d696   21 hours ago   1.06GB
openwrt/imagebuilder   x86-64-v22.03.4           09e27754d696   21 hours ago   1.06GB
```

fantom-x, 2023-04-22:
Something changed in the last couple of days: `openwrt/imagebuilder:x86-64-v22.03.3` is fine, but `openwrt/imagebuilder:x86-64-v22.03.4` is building `master`

```
docker pull openwrt/imagebuilder:x86-64-v22.03.3
x86-64-v22.03.3: Pulling from openwrt/imagebuilder
7d66b83ec869: Already exists 
676c8bbc92ba: Already exists 
e885bd6adb79: Already exists 
e009cd5051ab: Already exists 
a0180050edf3: Already exists 
b4f9f5379efd: Already exists 
a64ac06c8193: Already exists 
4f4fb700ef54: Already exists 
79f1ea206871: Pull complete 
Digest: sha256:4842b4dfcbf73bc59566c00dfd0292c81426e4204505705ab36201a8f822c439
Status: Downloaded newer image for openwrt/imagebuilder:x86-64-v22.03.3
docker.io/openwrt/imagebuilder:x86-64-v22.03.3

docker container run -it openwrt/imagebuilder:x86-64-v22.03.3 cat repositories.conf
## Place your custom repositories here, they must match the architecture and version.
# src/gz %n https://downloads.openwrt.org/releases/22.03.3
# src custom file:///usr/src/openwrt/bin/x86/packages

## Remote package repositories
src/gz openwrt_core https://downloads.openwrt.org/releases/22.03.3/targets/x86/64/packages
src/gz openwrt_base https://downloads.openwrt.org/releases/22.03.3/packages/x86_64/base
src/gz openwrt_luci https://downloads.openwrt.org/releases/22.03.3/packages/x86_64/luci
src/gz openwrt_packages https://downloads.openwrt.org/releases/22.03.3/packages/x86_64/packages
src/gz openwrt_routing https://downloads.openwrt.org/releases/22.03.3/packages/x86_64/routing
src/gz openwrt_telephony https://downloads.openwrt.org/releases/22.03.3/packages/x86_64/telephony

## This is the local package repository, do not remove!
src imagebuilder file:packages

option check_signature
```

==============

```
docker pull openwrt/imagebuilder:x86-64-v22.03.4
x86-64-v22.03.4: Pulling from openwrt/imagebuilder
7d66b83ec869: Already exists 
676c8bbc92ba: Already exists 
e885bd6adb79: Already exists 
e009cd5051ab: Already exists 
a0180050edf3: Already exists 
b4f9f5379efd: Already exists 
a64ac06c8193: Already exists 
4f4fb700ef54: Already exists 
761cea502bda: Pull complete 
Digest: sha256:309f76b79179da2cf6058a42ba7ea962216c9eb95dd501bea89ed50913ca3e1f
Status: Downloaded newer image for openwrt/imagebuilder:x86-64-v22.03.4
docker.io/openwrt/imagebuilder:x86-64-v22.03.4

docker container run -it openwrt/imagebuilder:x86-64-v22.03.4 cat repositories.conf
## Place your custom repositories here, they must match the architecture and version.
# src/gz %n https://downloads.openwrt.org/snapshots
# src custom file:///usr/src/openwrt/bin/x86/packages

## Remote package repositories
src/gz openwrt_core https://downloads.openwrt.org/snapshots/targets/x86/64/packages
src/gz openwrt_base https://downloads.openwrt.org/snapshots/packages/x86_64/base
src/gz openwrt_kmods https://downloads.openwrt.org/snapshots/targets/x86/64/kmods/5.15.107-1-e469f5589b4c7b368924a6e4f8f7407f
src/gz openwrt_luci https://downloads.openwrt.org/snapshots/packages/x86_64/luci
src/gz openwrt_packages https://downloads.openwrt.org/snapshots/packages/x86_64/packages
src/gz openwrt_routing https://downloads.openwrt.org/snapshots/packages/x86_64/routing
src/gz openwrt_telephony https://downloads.openwrt.org/snapshots/packages/x86_64/telephony

## This is the local package repository, do not remove!
src imagebuilder file:packages

option check_signature
```

aparcar, 2023-04-22:
Thanks for reporting. I'll check what's wrong

fantom-x, 2023-04-23:
@aparcar , can you also explain the tagging scheme on the overview page https://hub.docker.com/r/openwrt/imagebuilder ? `x86-64-22.03.4-SNAPSHOT` is a confusing name.

aparcar, 2023-04-23:
Could you please check again?

fantom-x, 2023-04-23:
It is working now, thx.

```
REPOSITORY             TAG               IMAGE ID       CREATED          SIZE
openwrt/imagebuilder   x86-64-v22.03.4   d21c83257557   43 minutes ago   1.05GB

=============================

Building package index...
Downloading https://downloads.openwrt.org/releases/22.03.4/targets/x86/64/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/22.03.4/targets/x86/64/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/x86_64/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/x86_64/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/x86_64/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/x86_64/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/x86_64/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/x86_64/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/x86_64/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/x86_64/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/x86_64/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/x86_64/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
```

aparcar, 2023-04-24:
Great thanks for reporting and testing :)

closed 2023-04-24 by aparcar