openwrt/docker #112: Cross architecture rootfs images are declared as amd64

bwoodsend, 2023-04-22:
```bash
> docker pull openwrtorg/rootfs:mips_24kc
mips_24kc: Pulling from openwrtorg/rootfs
Digest: sha256:44e17879beb8da4b64aad5569fbe550ec7320f0f390202758db5026b492493cb
Status: Image is up to date for openwrtorg/rootfs:mips_24kc
docker.io/openwrtorg/rootfs:mips_24kc

> docker image inspect openwrtorg/rootfs:mips_24kc | jq '.[].Architecture'
"amd64"
```
This used to not matter because Docker could be made to run it by setting the architecture explicitly using `--platform` but now that's not allowed.
```bash
> docker run -it --rm --platform=linux/mips_24kc openwrtorg/rootfs:mips_24kc
Unable to find image 'openwrtorg/rootfs:mips_24kc' locally
mips_24kc: Pulling from openwrtorg/rootfs
Digest: sha256:44e17879beb8da4b64aad5569fbe550ec7320f0f390202758db5026b492493cb
Status: Image is up to date for openwrtorg/rootfs:mips_24kc
docker: Error response from daemon: image with reference openwrtorg/rootfs:mips_24kc was found but does not match the specified platform: wanted linux/mips_24kc, actual: linux/amd64.
```
I suspect you need to set the `platforms` option [here](https://github.com/openwrt/docker/blob/3598be3ba023db867f8003f4dd7012d957a6cce5/.github/workflows/containers.yml#L197).

aparcar, 2023-04-23:
Looks like you're already working on a PR to fix that, thanks!

bwoodsend, 2023-04-23:
Hmm, almost. But MIPS is causing issues.

I added [a dumb *run it* test](https://github.com/bwoodsend/docker/blob/9f32de7a3338db99170066f9952eaef540eedaad/.github/workflows/containers.yml#L207) which sure enough [reproduces](https://github.com/bwoodsend/docker/actions/runs/4778437304) this ticket. Adding [my hypothesised one line fix](https://github.com/bwoodsend/docker/blob/854497592929395b5bf841c782468b20b395b805/.github/workflows/containers.yml#L207) fixes every platform [except MIPS](https://github.com/bwoodsend/docker/actions/runs/4777666699/jobs/8493558779) which is raising Docker's not so informative *wrong architecture* error. 

Running it locally gives me the same thing:
```fish
> docker build --platform linux/mips_24kc --build-arg 'DOWNLOAD_FILE=openwrt-.*-rootfs.tar.gz' --build-arg WORKDIR=/ --build-arg USER=root --build-arg VERSION_PATH=snapshots --build-arg TARGET=malta/be --build-arg BASE_IMAGE=scratch --build-arg CMD=ash .
[+] Building 2.6s (22/22) FINISHED                                                                                                                                                                                                            
 => [internal] load .dockerignore                                                                                                                                                                                                        0.0s
...
 => => writing image sha256:b3d596932ac4091b06535b990144d701814f26af08d573bb4484430b362005d3                                                                                                                                             0.0s
> docker run --platform=linux/mips_24kc sha256:b3d596932ac4091b06535b990144d701814f26af08d573bb4484430b362005d3 uname -m
exec /bin/uname: exec format error
```

I wasn't too sure what Docker/qemu aliases as `mips_24kc` (for the `--platform` flag) so I downloaded the rootfs tarball and tried to empirically determine which qemu emulator it needs:
```fish
wget https://downloads.openwrt.org/snapshots/targets/malta/be/openwrt-malta-be-default-rootfs.tar.gz
tar xf openwrt-malta-be-default-rootfs.tar.gz
for qemu in /bin/qemu-*static; echo $qemu; $qemu lib/libc.so; end
```
`qemu-mips-static` is the emulator that matches the architecture of the binaries in that tarball which corresponds to `--platform=linux/mips`:

```bash
> docker run --rm --privileged multiarch/qemu-user-static --reset -p yes --credential yes | grep mips
Setting /usr/bin/qemu-mips-static as binfmt interpreter for mips
Setting /usr/bin/qemu-mipsel-static as binfmt interpreter for mipsel
Setting /usr/bin/qemu-mipsn32-static as binfmt interpreter for mipsn32
Setting /usr/bin/qemu-mipsn32el-static as binfmt interpreter for mipsn32el
Setting /usr/bin/qemu-mips64-static as binfmt interpreter for mips64
Setting /usr/bin/qemu-mips64el-static as binfmt interpreter for mips64el
```

Setting `--platform=linux/mips` also doesn't work though:
```fish
> docker build --platform linux/mips -q --build-arg 'DOWNLOAD_FILE=openwrt-.*-rootfs.tar.gz' --build-arg WORKDIR=/ --build-arg USER=root --build-arg VERSION_PATH=snapshots --build-arg TARGET=malta/be --build-arg BASE_IMAGE=scratch --build-arg CMD=ash .
sha256:f8e5382d05afe40ea1618ba2dad3f1b2b1b3dd62a5dc744851b304682d465fbc
> docker run --platform=linux/mips sha256:f8e5382d05afe40ea1618ba2dad3f1b2b1b3dd62a5dc744851b304682d465fbc uname -m
exec /bin/uname: exec format error
```

I suspect that this reveals the cause:
```fish
> docker run --platform=linux/mips sha256:f8e5382d05afe40ea1618ba2dad3f1b2b1b3dd62a5dc744851b304682d465fbc /lib/libc.so
qemu-mipsn32-static: /lib/libc.so: Invalid ELF image for this architecture
```
Docker is using `qemu-mipsn32-static` instead of `qemu-mips-static`.

So unless you have any ideas, I think the best I can do is fix the others and disable MIPs rootfs then poke around the Docker community to see why Docker is using the wrong qemu emulator.

aparcar, 2023-04-23:
@dangowrt do you have an idea on how to treat this?

aparcar, 2023-04-23:
Thank you for the insights, very helpful! I'd be fine to drop it for now

PR #113 (2023-05-25): Fix multiarch rootfs containers being mislabelled as AMD64.
* Set the platforms argument to docker/build-push-action.
* Insert a minimal *is it runable* test in between building and pushing the rootfs images.
* <s>Disable building rootfs for malta/be mips_24kc due to the images being unrunable (see https://github.com/openwrt/docker/issues/112#issuecomment-1519099490).</s> Not any more - they're working now.

Fixes #112.
commit e61e5180 (2023-05-25): Fix multiarch rootfs containers being mislabelled as AMD64.

* Set the platforms argument to docker/build-push-action.

* Insert a minimal *is it runable* test in between building and pushing
  the rootfs images.

Fixes #112.
closed 2023-05-25 by aparcar