openwrt/docker #118: Building `libnl` with the sdk fails

Sohn123, 2023-07-13:
Hi,

we tried using the sdk container to build `libnl`. We tried different versions: `latest`, `mvebu-cortexa9-22.03.3`, mvebu-cortexa9-22.03.4` and `mvebu-cortexa9-22.03.5`. We executed the following commands:

```
./scripts/feeds update -a
make defconfig
./scripts/feeds install libnl
make V=sc package/libnl/compile
```

The build gets stuck for several minutes with the following message:

```
make[1]: Entering directory '/builder'
make[2]: Entering directory '/builder/package/toolchain'
Makefile:735: WARNING: skipping libgomp -- package has no install section
mkdir -p /builder/bin/targets/x86/64/packages /builder/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc/CONTROL /builder/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /builder/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc/lib
cp -fpR /builder/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/libgcc_s.so.* /builder/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc/lib/
find /builder/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/builder/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/builder/scripts/strip-kmod.sh" PATCHELF="/builder/staging_dir/host/bin/patchelf" /builder/scripts/rstrip.sh /builder/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc
rstrip.sh: /builder/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc/lib/libgcc_s.so.1: shared object
(cd /builder/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@";
) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;
)
install -d -m0755 /builder/bin/targets/x86/64/packages
/builder/staging_dir/host/bin/fakeroot /builder/staging_dir/host/bin/bash /builder/scripts/ipkg-build -m
"" /builder/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc /builder/bin/targets/x86/64/packages
```

Does anyone know what we are doing wrong, or why it tries to compile the toolchain at all? I thought the sdk contains a precompiled toolchain.

rgwohlbold, 2023-07-14:
It seems like fakeroot hangs, using `--ulimit nofile=1024:524288` as an option to `docker run` fixes the issue

Sohn123, 2023-07-14:
Probably the problem with the `ulimits` happen because the user inside the container is now `buildbot` instead of `root`. Maybe a hint in the README would be helpful, that if `fakeroot` hangs changing the `ulimits` is required.