openwrt/docker #125: Proper documentation request

Unb0rn, 2023-10-17:
I'm trying to use the SDK image to build kmod for my target and I'm totally lost - I can't understand whether it's an image or documentation issue.
I'm trying to build for Xiaomi AX9000 it has `ipq807x` target with `generic` subtarget. What I tried to do:

1. Pulled an image `openwrt/sdk:ipq807x-generic-v23.05.0` and ran it as usual.
2. Updated the feeds `./scripts/feeds update -a` and after exploring a bit discovered that this v23.05.0 image builds 23.05 branch instead of tag that has commit hash 6637af95aa (at least the official build for my device) Is it okay?
3. Tried to create .config and that's where all the problems started. Running `make menuconfig` starts the configuration interface, however it's impossible to change target in it. If I save it "as is" and open the resulting .config, I can see that it has a weird default target `CONFIG_TARGET_BOARD=bcm27xx` (is it RPi 3?) This target is aarch64-cortexa53 too but it's not ipq807x obviously. Leaving the target as is produces the kmod that I obviously can't install.

Am I doing something wrong? Is it an image issue? Or maybe SDK problem? I'm new to OpenWrt build system and I definitely don't want to assemble build environment on the host.

Unb0rn, 2023-10-18:
Turns out it was the image problem - rebuilding the image manually with proper SDK solves the issue. I think this is an issue now)

aparcar, 2024-02-05:
Multiple targets share the same architecture. I tried with the image you mentioned and it works just fine, the packages created are `aarch64_cortex`.

closed 2024-02-05 by aparcar