openwrt/docker #141: push linux/arm64 images to dockerhub too

itay-sho, 2024-07-20:
I have an ARM64 host
looked on Dockerhub and saw that only linux/amd64 images are being pushed.

Is it possible to build the images to linux/arm64 as well?
In case there are some missing parts I can try to contribute.

Thank you

AreYouLoco, 2025-04-28:
I am giving +1 to this request. I bumped into same problem.

Wouldnt like to trash my host system so I prefer docker to isolate stuff.

I guess the problem is that SDK tar.gz on dowbloads.openwrt.org are only x64 and thats hard-coded in Dockerfile.

AreYouLoco, 2025-04-28:
So actually needs to be rewuested there first so there is something to refer to

AreYouLoco, 2025-04-28:
And in the mean time I will try with self-build image.