openwrt/docker #148: No <arch> tags for SDK?

bam80, 2024-09-28:
Why do we have SDK tags in the _\<target\>-\<subtarget\>_ format?
https://github.com/openwrt/docker/blob/7d14bdaa7aa194f1b26767e07ee3857d30f000ed/README.md?plain=1#L53
Aren't may of those combinations correspond to a single _\<arch\>_-specific SDK?

ynezz, 2024-09-29:
Actually both formats are supported and available, can you double check that and eventually please fix the documentation via pull request? Thanks!

bam80, 2024-09-29:
Still, what is the purpose of having <target>-<subtarget> SDKs?
Are they all belongings to a single <arch> actually the same, or there is something different between them, rather than name?

ynezz, 2024-10-01:
> Still, what is the purpose of having <target>-<subtarget> SDKs?

Convenience? 

> or there is something different between them, rather than name?

They're just cheap tags, having same content, indeed:

```shell
$ skopeo inspect --no-tags docker://ghcr.io/openwrt/sdk:armsr-armv8 | jq -r '.Digest'
sha256:8908ee25587ae258721206e04d27def76436707a7965e51d8b2c8c7af454a814

$skopeo inspect --no-tags docker://ghcr.io/openwrt/sdk:aarch64_generic | jq -r '.Digest'
sha256:8908ee25587ae258721206e04d27def76436707a7965e51d8b2c8c7af454a814
```

closed 2024-10-01 by aparcar

bam80, 2024-10-01:
> Convenience?
> They're just cheap tags, having same content, indeed.

Thanks for the clarifying.

Do we really need this? For me, it was just a confusion.
I wound end up with just \<arch\> tags for SDK, actually.
Thoughts?

PR #149 (2024-10-01): Update README.md: add missing SDK tags
fixes #148

As @ynezz noticed, SDKs with the missing tags are actually present, so add them to the doc.
commit 00a38e53 (2024-10-01): Update README.md: add missing SDK tags

fix #148