openwrt/docker #155: gpg: Can't check signature: No public key

moetayuko, 2024-11-12:
Experienced the following issue with `mipsel_74kc-openwrt-23.05`, `mipsel_mips32-openwrt-23.05`
 and `arm_cortex-a9-openwrt-23.05` docker images.
```
--2024-11-12 01:39:47--  https://downloads.openwrt.org/releases/23.05-SNAPSHOT/targets/bcm53xx/generic/sha256sums
Resolving downloads.openwrt.org (downloads.openwrt.org)... 146.75.78.132, 2a04:4e42:83::644
Connecting to downloads.openwrt.org (downloads.openwrt.org)|146.75.78.132|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 128475 (125K) [text/plain]
Saving to: ‘sha256sums’

     0K .......... .......... .......... .......... .......... 39% 25.5M 0s
    50K .......... .......... .......... .......... .......... 79% 33.1M 0s
   100K .......... .......... .....                           100%  226M=0.003s

2024-11-12 01:39:48 (35.0 MB/s) - ‘sha256sums’ saved [128475/128475]

--2024-11-12 01:39:48--  https://downloads.openwrt.org/releases/23.05-SNAPSHOT/targets/bcm53xx/generic/sha256sums.asc
Resolving downloads.openwrt.org (downloads.openwrt.org)... 146.75.78.132, 2a04:4e42:83::644
Connecting to downloads.openwrt.org (downloads.openwrt.org)|146.75.78.132|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 269 [text/plain]
Saving to: ‘sha256sums.asc’

     0K                                                       100% 24.0M=0s

2024-11-12 01:39:48 (24.0 MB/s) - ‘sha256sums.asc’ saved [269/269]

gpg: Signature made Wed 06 Nov 2024 12:59:04 PM UTC
gpg:                using EDDSA key 92C561DE55AE6552F3C736B82B0151090606D1D9
gpg: Can't check signature: No public key
```

ynezz, 2024-11-12:
The signing keys were changed https://lists.openwrt.org/pipermail/openwrt-devel/2024-November/043358.html (snapshot/24.10) and https://lists.openwrt.org/pipermail/openwrt-devel/2024-November/043358.html (23.05), so until the new containers are provided with the new key https://github.com/openwrt/docker/commit/954a9ef5c666b7407d074ddabffc23769f585bc4, you can use following workaround:

```shell
gpg --recv-keys 0x1D53D1877742E911
./setup.sh
```

closed 2024-11-12 by ynezz

muink, 2024-12-22:
Sames issue in `aarch64_cortex-a53-SNAPSHOT` docker images.
https://github.com/fantastic-packages/packages/actions/runs/12451377062/job/34759249713#step:14:1630
```
  --2024-12-22 04:57:12--  https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/sha256sums
  Resolving downloads.openwrt.org (downloads.openwrt.org)... 146.75.78.132, 2a04:4e42:83::644
  Connecting to downloads.openwrt.org (downloads.openwrt.org)|146.75.78.132|:443... connected.
  HTTP request sent, awaiting response... 200 OK
  Length: 780675 (762K) [text/plain]
  Saving to: ‘sha256sums’
  
       0K .......... .......... .......... .......... ..........  6% 30.5M 0s
     750K .......... ..                                         100%  188M=0.008s
  
  2024-12-22 04:57:13 (98.9 MB/s) - ‘sha256sums’ saved [780675/780675]
  
  --2024-12-22 04:57:13--  https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/sha256sums.asc
  Resolving downloads.openwrt.org (downloads.openwrt.org)... 146.75.78.132, 2a04:4e42:83::644
  Connecting to downloads.openwrt.org (downloads.openwrt.org)|146.75.78.132|:443... connected.
  HTTP request sent, awaiting response... 200 OK
  Length: 299 [text/plain]
  Saving to: ‘sha256sums.asc’
  
       0K                                                       100% 10.8M=0s
  
  2024-12-22 04:57:13 (10.8 MB/s) - ‘sha256sums.asc’ saved [299/299]
  
  gpg: Signature made Sun 22 Dec 2024 04:23:51 AM UTC
  gpg:                using EDDSA key 92C561DE55AE6552F3C736B82B0151090606D1D9
  gpg: BAD signature from "OpenWrt Build System (Nitrokey3) <contact@openwrt.org>" [unknown]
```

ynezz, 2024-12-22:
> Sames issue in aarch64_cortex-a53-SNAPSHOT docker images.

No, its different issue:

>   gpg: BAD signature from "OpenWrt Build System (Nitrokey3) <contact@openwrt.org>" [unknown]

This means a problem, cryptographic signature of `sha256sums` stored in `sha256sums.asc` is invalid. Probably another variation of CDN de-sync issues? https://github.com/openwrt/openwrt/issues/16511