openwrt/docker #167: rootfs generation changes cause setup.sh to fail

efahl, 2025-04-04:
@ynezz https://github.com/openwrt/docker/commit/79999f3c7e586ef97ddc17d1665a1b3956efc102 is causing an issue with at least mediatek/filogic.  (Chased it down from https://forum.openwrt.org/t/the-openwrt-firmware-selector/81721/1302 )

```
host$ podman run -ti ghcr.io/openwrt/imagebuilder:mediatek-filogic-master
```
Then inside container run the setup.sh, it hits line 13, where the `&& rm -rf /builder/keys/` fails with "Permission denied" and kills the container.

```
$ . setup.sh
--2025-04-04 17:06:16--  https://downloads.openwrt.org/snapshots/targets/mediatek/filogic/sha256sums
Resolving downloads.openwrt.org (downloads.openwrt.org)... 151.101.130.132, 151.101.194.132, 151.101.2.132, ...
Connecting to downloads.openwrt.org (downloads.openwrt.org)|151.101.130.132|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 5351880 (5.1M) [text/plain]
Saving to: ‘sha256sums’

sha256sums                                        100%[=============================================================================================================>]   5.10M  30.2MB/s    in 0.2s

2025-04-04 17:06:16 (30.2 MB/s) - ‘sha256sums’ saved [5351880/5351880]

--2025-04-04 17:06:16--  https://downloads.openwrt.org/snapshots/targets/mediatek/filogic/sha256sums.asc
Resolving downloads.openwrt.org (downloads.openwrt.org)... 151.101.194.132, 151.101.66.132, 151.101.2.132, ...
Connecting to downloads.openwrt.org (downloads.openwrt.org)|151.101.194.132|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 299 [text/plain]
Saving to: ‘sha256sums.asc’

sha256sums.asc                                    100%[=============================================================================================================>]     299  --.-KB/s    in 0s

2025-04-04 17:06:16 (45.3 MB/s) - ‘sha256sums.asc’ saved [299/299]

gpg: directory '/builder/.gnupg' created
gpg: keybox '/builder/.gnupg/pubring.kbx' created
gpg: key 1D53D1877742E911: 3 signatures not checked due to missing keys
gpg: /builder/.gnupg/trustdb.gpg: trustdb created
gpg: key 1D53D1877742E911: public key "OpenWrt Build System (Nitrokey3) <contact@openwrt.org>" imported
gpg: key 28A39BC32074BE7A: 2 signatures not checked due to missing keys
gpg: key 28A39BC32074BE7A: public key "OpenWrt Build System (PGP key for 19.07 release builds) <pgpsign-19.07@openwrt.org>" imported
gpg: key CD84BCED626471F1: 2 signatures not checked due to missing keys
gpg: key CD84BCED626471F1: public key "OpenWrt Build System (PGP key for unattended snapshot builds) <pgpsign-snapshots@openwrt.org>" imported
gpg: key 88CA59E88F681580: public key "OpenWrt Build System (PGP key for 21.02 release builds) <pgpsign-21.02@openwrt.org>" imported
gpg: key CD54E82DADB3684D: public key "OpenWrt Build System (GnuPGP key for 22.03 release builds) <pgpsign-22.03@openwrt.org>" imported
gpg: Total number processed: 5
gpg:               imported: 5
gpg: no ultimately trusted keys found
rm: cannot remove '/builder/keys/CD54E82DADB3684D.asc': Permission denied
rm: cannot remove '/builder/keys/626471F1.asc': Permission denied
rm: cannot remove '/builder/keys/2074BE7A.asc': Permission denied
rm: cannot remove '/builder/keys/0x1D53D1877742E911.asc': Permission denied
rm: cannot remove '/builder/keys/88CA59E8.asc': Permission denied
<bam dropped back to host>
```

efahl, 2025-04-04:
More info, owner is root:
```
buildbot@a136411359ab:~$ whoami
buildbot

buildbot@a136411359ab:~$ ls -la /builder/keys/
total 32
drwxr-xr-x 2 root     root     4096 Apr  3 09:13 .
drwxr-xr-x 4 buildbot buildbot 4096 Apr  4 17:12 ..
-rw-r--r-- 1 root     root     3139 Apr  3 09:13 0x1D53D1877742E911.asc
-rw-r--r-- 1 root     root     3294 Apr  3 09:13 2074BE7A.asc
-rw-r--r-- 1 root     root     6400 Apr  3 09:13 626471F1.asc
-rw-r--r-- 1 root     root     3216 Apr  3 09:13 88CA59E8.asc
-rw-r--r-- 1 root     root     3988 Apr  3 09:13 CD54E82DADB3684D.asc

buildbot@a136411359ab:~$ rm -rf /builder/keys/
rm: cannot remove '/builder/keys/CD54E82DADB3684D.asc': Permission denied
rm: cannot remove '/builder/keys/626471F1.asc': Permission denied
rm: cannot remove '/builder/keys/2074BE7A.asc': Permission denied
rm: cannot remove '/builder/keys/0x1D53D1877742E911.asc': Permission denied
rm: cannot remove '/builder/keys/88CA59E8.asc': Permission denied
```

efahl, 2025-04-04:
Aha, just have to wait for it to filter through.  Got it!

ynezz, 2025-04-05:
> podman run -ti ghcr.io/openwrt/imagebuilder:mediatek-filogic-master

Looking at that container metadata, its indeed some problem somewhere, because I've triggered rebuild of both `main` and `openwrt-24.10` images yesterday, but this container still has old, unfixed content.

```
"org.opencontainers.image.created": "2025-04-03T09:13:39.851Z",
"org.opencontainers.image.revision": "e8012cce6abbfe3cd0a5bbc8edb25717dd385fde",
```

I've triggered a rebuild of `main` again.

ynezz, 2025-04-05:
Indeed, the Push stage runs, but the actual push to the registries doesn't happen, it broke with cda8065dc5d8.

closed 2025-04-05 by ynezz

ynezz, 2025-04-05:
Should be fixed, thanks for the report.

commit 1c462828 (2025-04-05): ci: fix pushing of ImageBuilder containers

Eric reported, that he is having problem with
imagebuilder:mediatek-filogic-master. The issue is caused by refactoring
of Build Push stage in commit cda8065dc5d8 ("ci: add smoke testing of
SDK and IB containers"), where I forget to enable pushing with `push:
true` input as the default is `push: false`.

So lets fix it by explicitly enabling the push input, while at it unify
the Push stage configs.

Fixes: #167
Fixes: cda8065dc5d8 ("ci: add smoke testing of SDK and IB containers")
Link: https://github.com/openwrt/docker/pull/168
Reported-by: Eric Fahl [... cut]
PR #168 (2025-04-05): ci: fix pushing of ImageBuilder containers
Eric reported, that he is having problem with
imagebuilder:mediatek-filogic-master. The issue is caused by refactoring of Build Push stage in commit cda8065dc5d8 ("ci: add smoke testing of SDK and IB containers"), where I forget to enable pushing with `push: true` input as the default is `push: false`.

So lets fix it by explicitly enabling the push input, while at it unify the Push stage configs.

Fixes: #167
Fixes: cda8065dc5d8 ("ci: add smoke testing of SDK and IB containers")
Link: https://github.com/openwrt/docker/pull/168
Reported-by: Eric Fahlg [... cut]