openwrt/docker #172: GPG keys permission issues during container setup

efahl, 2025-04-22:
@ynezz 
https://github.com/openwrt/docker/commit/648a9bd5ba25e02c53995e95401b77734c46a1eb

Seeing this still on the lantiq build...  The build-at date for the container is well after the merge (2025-04-09T20:33:48Z ~66 hours ago), so it seems like it should have picked up the changes.

```
$ podman run -ti ghcr.io/openwrt/imagebuilder:lantiq-xrx200-master

buildbot@cde0e1afd7b8:~$ ls -la
total 5176
drwxr-xr-x 4 buildbot buildbot    4096 Apr 22 20:10 .
dr-xr-xr-x 1 root     root        4096 Apr 22 20:09 ..
-rw-r--r-- 1 buildbot buildbot     220 Mar 27  2022 .bash_logout
-rw-r--r-- 1 buildbot buildbot    3526 Mar 27  2022 .bashrc
drwx------ 3 buildbot buildbot    4096 Apr 22 20:10 .gnupg
drwxr-xr-x 2 root     root        4096 Apr  3 05:09 keys      <<<<<<<< still root
-rw-r--r-- 1 buildbot buildbot     807 Mar 27  2022 .profile
-rwxr-xr-x 1 root     root         989 Apr  3 05:08 setup.sh
-rw-r--r-- 1 buildbot buildbot 5262079 Apr 22 05:49 sha256sums
-rw-r--r-- 1 buildbot buildbot     299 Apr 22 05:49 sha256sums.asc

buildbot@cde0e1afd7b8:~$ ls -la keys/
total 32
drwxr-xr-x 2 root     root     4096 Apr  3 05:09 .
drwxr-xr-x 4 buildbot buildbot 4096 Apr 22 20:10 ..
-rw-r--r-- 1 root     root     3139 Apr  3 05:08 0x1D53D1877742E911.asc
-rw-r--r-- 1 root     root     3294 Apr  3 05:08 2074BE7A.asc
-rw-r--r-- 1 root     root     6400 Apr  3 05:08 626471F1.asc
-rw-r--r-- 1 root     root     3216 Apr  3 05:08 88CA59E8.asc
-rw-r--r-- 1 root     root     3988 Apr  3 05:08 CD54E82DADB3684D.asc

buildbot@cde0e1afd7b8:~$ ./setup.sh
...
rm: cannot remove '/builder/keys/CD54E82DADB3684D.asc': Permission denied
rm: cannot remove '/builder/keys/626471F1.asc': Permission denied
rm: cannot remove '/builder/keys/2074BE7A.asc': Permission denied
rm: cannot remove '/builder/keys/0x1D53D1877742E911.asc': Permission denied
rm: cannot remove '/builder/keys/88CA59E8.asc': Permission denied
```

efahl, 2025-04-22:
Oh, wait!  Is that the build-at date for the imagebuilder?  I think so, so maybe the docker container itself is ancient and just needs to be rebuilt.

ynezz, 2025-04-23:
IB needs to be fixed https://github.com/openwrt/openwrt/issues/18410

efahl, 2025-05-09:
Should be fixed by https://github.com/openwrt/openwrt/commit/b88f3ebb3341281fa7e61dc4f039062e72c86b9e

closed 2025-05-09 by efahl