openwrt/docker #184: 23.05.6: missing containers for all targets

Ro-1-2-3, 2025-08-26:
Hello,

It’s been a week since the release of 23.05.6, but none of the imagebuilder containers for its targets had yet been created (with the exception of ramips/mt7621 target, which was created manually, as a response to somebody’s open issue for that particular target).

For example, the mediatek-filogic target container, which I need, is missing, so I’ve been getting errors when trying to build a 23.05.6 image with my custom package selection, both via router ASU and via firmware selector website.

Would be nice if this issue is addressed and fixed.

Thank you,

R

efahl, 2025-08-27:
Details at https://github.com/openwrt/docker/issues/182#issuecomment-3224478911

Ro-1-2-3, 2025-08-29:
This was fixed. Thank you!